automote/ESP-CoAP #5: Using ESP8266 as Server and trying to communicate with another server

srinadhpotnuru, 2017-09-05:
Hi,
I am trying to use ESP8266 as a server and to connect it to another server for M2M event based communication. But while including both libraries of coap_server.h and coap_client.h it is not compiling and showing the following error - 
"ESP01_COAP_SERVER:9: error: 'coapClient' does not name a type

 coapClient coap_Client;

 ^

exit status 1
'coapClient' does not name a type"

Is it possible for coap_server instance to generate GET, POST requests

lovelesh, 2017-11-07:
@srinadhpotnuru coap_server will not be able to generate GET, POST request, but with small tweak in the source code, you will be able to make it work.
I would not suggest to add this functionality as server should only serve the requests, but you can use a combination of coap server and client to cater to your requirement

E12351, 2017-11-14:
@lovelesh In the industry level instruments, there is this functionality that server can send GET and POST methods. I'm working on one of the water meters that has this functionality. To communicate with the water meter we need act as both server and client.

lovelesh, 2021-06-05:
@E12351 the easy way is to write a small client code inside the server. Logically servers are not designed to make new connections to other devices.

closed 2021-06-05 by lovelesh