automote/ESP-CoAP #15: #Define BUF MAX SIZE 250

ebleme, 2018-10-11:
Hi. I am using this library to test CoAP performance for various situations. But I can't send CoAP packets if their payload length bigger than 250 characters.

I think the problem is on that lines 

**coap_client.h**
`#define BUF_MAX_SIZE 250`

**coap_client.cpp   inside the coapClient::sendPacket method**

`
        // make payload
	if (packet.payloadlen > 0) {

		if ((packetsize + 1 + packet.payloadlen) >= buf_max_size) {
			return 0;
		}
		*p++ = 0xFF;
		memcpy(p, packet.payload, packet.payloadlen);
		packetSize += 1 + packet.payloadlen;
	}
`

What can I do to make this library support more than 250 length of payload since -as much as I know- CoAP supports 1024 bytes of payload. 

Thanks.

abcosmi, 2018-10-12:
Hi @ebleme . 

I made the following changes on the code to send 1000 bytes on the client side:

**coap_client.h**:

Set BUF_MAX_SIZE to 1024.
Change the type of payloadlen attribute of coapPacket class from **uint8_t** to **uint16_t**;

> //coap packet class
class coapPacket {
	public:
		uint8_t type;
		uint8_t code;
		uint8_t *token;
		uint8_t tokenlen;
		uint8_t *payload;
		**uint16_t payloadlen;   <------**
		uint16_t messageid;
		uint8_t optionnum;
		coapOption options[MAX_OPTION_NUM];
};

In my case, I've needed to select another lwIP variant on Arduino IDE.

If you click on **Tools** and **lwIP Variant**, select **v1.4 Higher Bandwidth** option.

I've tested this using another ESP as coap server using the same configs.

I don't know if it's the right way, but it's my workaround.

lovelesh, 2018-10-12:
@abcosmi. Thanks for bringing it to notice. If you like you can send a PR. i will merge into the master code

ebleme, 2018-10-12:
@abcosmi Thank you very much. I made the changes and it solved the problem. I can now send bigger datas.